Gold872/elastic_dashboard #11: Graphing does not have an X axis

daltzctr, 2023-10-30:
[image]

Gold872, 2023-10-30:
This is because of the way the graph is being rendered. It's using a ChartSeriesController to improve performance when there are a lot of widgets. This essentially has the X axis continuously increasing, and there is not a way of it to display time. If I wanted to include numbers on the X axis, it would severely slow the program down if graphs are being rendered.

daltzctr, 2023-10-30:
The graph is not very usable without the X axis.

Gold872, 2023-10-30:
The graph is really intended to see the general trend in how data is changing, as during matches you won't want to be looking at precise timing because there's a robot to operate. For applications where you need to precisely see things over time, I recommend using Glass's plotting feature instead.

closed 2023-11-07 by Gold872