Gold872/elastic_dashboard #12: Graph options sometimes don't set

daltzctr, 2023-10-30:
Looks like I have to add or remove any widget to "force" it to update it's settings.

Gold872, 2023-10-30:
Could you send a video of this happening?

daltzctr, 2023-10-30:
https://github.com/Gold872/elastic-dashboard/assets/105223895/eed0d5e5-5ad2-483b-a869-460df8c4058b

Gold872, 2023-10-30:
You have to hit enter before clicking "Close". From what I know this is a limitation of Flutter so I don't think I can fix this.

daltzctr, 2023-10-30:
Can there be a "confirm" or "save" button?

Gold872, 2023-10-30:
I'll think about a way of doing this since it's a good idea

Gold872, 2023-11-25:
I finally figured out the solution to this. It had to do with the graph widget being a stateful widget that wasn't in a stream builder. The solution was very convoluted but the widget should now successfully rebuild when you change its settings.

closed 2023-11-25 by Gold872