Gold872/elastic_dashboard #16: Dashboard is frozen after tabbing back in after awhile

daltzctr, 2023-10-30:
Had it open for maybe 30 minutes, tabbed away. Tabbed back in and it's completely locked up. 

https://github.com/Gold872/elastic-dashboard/assets/105223895/5c6a5963-005c-439b-9789-3fb3461593d6

Gold872, 2023-10-30:
Our team has had similar issues like this, I believe pressing something on your keyboard should fix it

Gold872, 2023-12-02:
I actually had this issue happen the other day, what fixed it was dragging the window which unfroze it. I haven't been able to find any consistency is when it is happening but I have a feeling it has something to do with the OS putting it in "sleep mode" to save power.

Let me know if you find a way of consistently reproducing this issue, but right now i'll try implementing things that could potentially fix it

J-Barta, 2024-08-31:
I know this is a while after, but I've had some users of my other flutter projects make similar comments about inexplicable freezes. They weren't able to find any rhyme or reason to it. It seems like it's just an occasional thing that occurs with Flutter applications.