Gold872/elastic_dashboard #9: Application refuses to launch if shared-preferences.json is corrupted

daltzctr, 2023-10-30:


daltzctr, 2023-10-30:
Located at `appdata/com.gold872/`

Gold872, 2023-10-30:
How exactly is the shared preferences file getting corrupted? That falls a bit more into an issue with your hard drive

daltzctr, 2023-10-30:
It's actually fairly common for json configs to be a point of error in programs. Task managering the program after it hard freezes in the middle of a write can corrupt the json.

If it fails to read, I expect it to recreate the json.

Gold872, 2023-10-31:
Since the shared preferences is handled by a library that is not maintained by me, this is something that you will have to suggest to the maintainers of the SharedPreferences plugin.

https://pub.dev/packages/shared_preferences
https://github.com/flutter/packages/tree/main/packages/shared_preferences

daltzctr, 2023-10-31:
Possible workaround you could do here https://github.com/flutter/flutter/issues/89211#issuecomment-915096452

closed 2023-11-07 by Gold872