Gold872/elastic_dashboard #19: Missing Widgets from Shuffleboard and SmartDashboard

Gold872, 2023-11-12:
- [ ]  3-Axis Accelerometer
- [ ] Accelerometer
- [ ] Encoder
- [ ] Motor Controller
- [ ] PID Command
- [ ] Profiled PID Controller
- [ ] Relay
- [ ] Ultrasonic
- [ ] Compressor
- [ ] Double Solenoid
- [ ] Solenoid
- [ ] Digital Input
- [ ] Digital Output
- [ ] Nidec Brushless

Gold872, 2023-11-16:
I know, but I want to have them in case if somebody wants to use it (a lot of times my team uses things like solenoids and compressors for debugging)

m10653, 2024-02-01:
Another big one is the Mechanism2d widget

NO-skcaj, 2024-05-19:
We're also missing an option for gyro visualization.

Gold872, 2025-06-25:
Closing this since most of these are either not needed in a driver dashboard or don't have any unique information to display outside a basic boolean indicator or text box