Gold872/elastic_dashboard #25: Graph widget not working

etiennebeaulac, 2024-01-07:
**Describe the bug**
When I add a Graph widget to the dashboard, a blank rectangle is rendered instead.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a number widget
2. Right click on it and change the widget to show as "Graph"
3. The inside of the widget is showing a blank rectangle
4. The same blank rectangle is shown in the widget parameters.

**Expected behavior**
The Graph widget is rendered and usable as in the docs.

**Screenshots**
[image]
[image]
[image]

**Desktop**
 - OS: Windows 10

Gold872, 2024-01-07:
Could you go into your log file (it's in `C:/Users/<user>/App data/Roaming/com.gold872/elastic_dashboard/`) and show me any errors that are there?

etiennebeaulac, 2024-01-07:
There is indeed an error that happens when the graph widget is added. The 1st is when it is initially added, the 2nd when I resized it, and the 3rd when I opened the widget properties.

```
[I]  [2024-01-07 13:30:48.483018] Starting application
[I]  Backup up shared_preferences.json to C:\Users\***\AppData\Roaming\com.gold872\elastic_dashboard\shared_preferences_backup.json
[I]  Configuring NT Widget Builder
[I]  Loading layout from json
[I]  Setting up shortcuts
[I]  Checking for updates
[I]  Network Tables connected on IP address localhost
[I]  Showing tab close confirmation for tab: Autonomous
[I]  Displaying add widget dialog
┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
│ LateInitializationError: Field 'lineWidth' has not been initialized.
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ #0   GraphWidget.build (package:elastic_dashboard/widgets/nt_widgets/single_topic/graph.dart)
│ #1   StatelessElement.build (package:flutter/src/widgets/framework.dart:5541)
│ #2   ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5471)
│ #3   Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
│ #4   ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:5453)
│ #5   ComponentElement.mount (package:flutter/src/widgets/framework.dart:5447)
│ #6   Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
│ #7   Element.updateChild (package:flutter/src/widgets/framework.dart:3831)
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ ⛔ Flutter Error
└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
│ LateInitializationError: Field 'lineWidth' has not been initialized.
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ #0   GraphWidget.build (package:elastic_dashboard/widgets/nt_widgets/single_topic/graph.dart)
│ #1   StatelessElement.build (package:flutter/src/widgets/framework.dart:5541)
│ #2   ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5471)
│ #3   Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
│ #4   ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:5453)
│ #5   ComponentElement.mount (package:flutter/src/widgets/framework.dart:5447)
│ #6   Element.inflateWidget (package:flutter/src/widgets/framework.dart:4326)
│ #7   Element.updateChild (package:flutter/src/widgets/framework.dart:3837)
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ ⛔ Flutter Error
└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
│ LateInitializationError: Field 'lineWidth' has not been initialized.
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ #0   GraphWidget.getEditProperties (package:elastic_dashboard/widgets/nt_widgets/single_topic/graph.dart)
│ #1   NTWidgetContainerModel.showEditProperties.<anonymous closure>.<anonymous closure> (package:elastic_dashboard/widgets/draggable_containers/models/nt_widget_container_model.dart:114)
│ #2   _StatefulBuilderState.build (package:flutter/src/widgets/basic.dart:7758)
│ #3   StatefulElement.build (package:flutter/src/widgets/framework.dart:5583)
│ #4   ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5471)
│ #5   StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:5634)
│ #6   Element.rebuild (package:flutter/src/widgets/framework.dart:5187)
│ #7   ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:5453)
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ ⛔ Flutter Error
└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

```

Gold872, 2024-01-07:
Oh I see what's going on, the variable for the line width doesn't get initialized when created, I'll fix this once I have access to my laptop and publish a new release

Sorry for the inconvenience!

etiennebeaulac, 2024-01-07:
No problem, thank you for your quick answer !

closed 2024-01-07 by Gold872