Gold872/elastic_dashboard #27: Dashboard issues

cherriae, 2024-01-27:
**Describe the bug**
A clear and concise description of what the bug is.

I have 2 issues here:
1. If you go to windows and change display scaling, the widgets don't resize to the size that screen is scaled to
- For example: I made layout on 100% scaling, then on my laptop its 125% scaling and the widgets are going across the dashboard window.
2. Whenever you make a new tab, I rename the tab to some value then I press enter but it doesn't close window but renames the tab to the value

**To Reproduce**
Steps to reproduce the behavior:
1. Scaling issues:
- Start by making layout in a certain scaling eg. 100%
- Then change to 125% scaling 
2. Tab rename window
- Create a new tab
- Rename tab
- Press enter button (keyboard) instead of pressing the button

**Expected behavior**
A clear and concise description of what you expected to happen.
1. I expected the layout to scale and widgets to be smaller
2. 

**Screenshots**
If applicable, add screenshots to help explain your problem.
(Will add some later)

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Firefox
 - Version: Latest

**Smartphone (please complete the following information):**
 - Device: Samsung
 - OS: Android 13
 - Browser: Chrome
 - Version: Latest

**Additional context**
Add any other context about the problem here.

N/A

Gold872, 2024-01-31:
> If you go to windows and change display scaling, the widgets don't resize to the size that screen is scaled to

This is standard for pretty much all windows apps, the scaling of the UI is entirely dependent on the screen scale you set on the OS, and there isn't a practical way of changing that behavior

> Whenever you make a new tab, I rename the tab to some value then I press enter on my keyboard but it doesn't close window but renames the tab to the value

This is intended. I do think it makes sense to have it auto close when pressing enter, but flutter's text submission is a bit weird with this as it doesn't tell you how the text editing finished, so there isn't an easy way of implementing this. As an alternative though hitting escape will do the same thing.

cherriae, 2024-01-31:
> This is standard for pretty much all windows apps, the scaling of the UI is entirely dependent on the screen scale you set on the OS, and there isn't a practical way of changing that behavior
>
>
> This [...]

I guess this isn't a problem on your side so I might as well close this issue.

closed 2024-01-31 by cherriae