Gold872/elastic_dashboard #28: Unable to Create tabs via Shuffleboard API

m10653, 2024-01-30:
**Describe the bug**
Utilizing the shuffleboard api and the getTab method fails to create a new tab. Where as the exact same code will cause a new tab to be created within shuffleboard dashboard itself.

**To Reproduce**
Steps to reproduce the behavior:
1. Run Shuffleboard.getTab("test");
2. Shuffleboard.selectTab("test");
3. Tab is created and switched to via Shuffleboard Dashboard
4. Elastic fails to do so.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Bug shown where tab Autonomous2 is not being created.
[image]

**Desktop (please complete the following information):**
 - OS:  Windows
 - WPILIb:  2024.1.1 - Java
 - Elastic  2024.0.2
 
**Additional context**
Add any other context about the problem here.

Gold872, 2024-01-31:
Elastic only creates a tab via the shuffleboard api when there's a new widget to be placed. I'll try experimenting with being able to create a tab when there's nothing on it, but I'm not sure if I will be able to get it reliably working. The shuffleboard schema on network tables is very finnicky and was a huge pain to get working consistently.

Gold872, 2024-01-31:
Feature has been added and will be released in the next update

closed 2024-01-31 by Gold872