Gold872/elastic_dashboard #29: Add ability to add raw network table Entries/paths Via Shuffleboard API

m10653, 2024-02-01:
Currently if you load load Elastic from a file you are able to display and view values directly out of network tables. However this is not possible to do programmatically via the Shuffleboard api. Where it is required to publish a second duplicate record of the data to network tables to be viewable within the dashboard. This is specifically a problem when wanting to view vision data or values from a co-processor publishing to network tables directly. Where the only way to programmatically publish these would be to read the data from network tables and republish it to shuffleboard on the robot adding unneeded latency.  

It would be nice to be able to just tell elastic to look at a particular network table entry/path by only publishing the string of the path itself vs needing to send a copy of the value.

Potently a non api breaking way to send this information over to elastic. 
```java
ShuffleboardTab tab = Shuffleboard.getTab("test");
tab.add("title", "nt-path", "/path/to/value");

```

Gold872, 2024-02-03:
This can already be done, but It's not really documented anywhere. Elastic stores the NT topic as a property, so it can be manually set with the Shuffleboard API via the widget's properties. To do this when adding a widget, you would use the `withProperties()` method.

Here's an example:
```java
Shuffleboard.getTab("Example Tab").add("Example Widget", 0.0) // 0.0 will never be displayed, so it doesn't really matter what you put for it
  .withProperties(Map.of("topic", "Whatever/Topic/You/Want/To/Display"));
```

Gold872, 2025-06-24:
Closing this since the Shuffleboard API has been removed for 2026