Gold872/elastic_dashboard #31: Elastic not loading layout json

vichik123, 2024-02-27:
We're trying to create a dashboard for our drivers, and we saved the layout in a [json](https://github.com/Gold872/elastic-dashboard/files/14418581/driver-dashboard-layout.json).
When we load the layout in elastic it just stops. Whether the simulation is running or not it doesn't load.
Did we forget something?

Gold872, 2024-02-27:
That's interesting, the JSON file looks valid. Could you get me the log of the program? It's in `%userprofile%/AppData/Roaming/com.gold872/elastic_dashboard/elastic-log.txt`

When there's an error loading a JSON it should send an alert notification but it looks like something might be going wrong with that.

When I have access to our shop I'll try loading this layout.

Gold872, 2024-02-28:
I was able to open it with my version just fine. Are both of your computers running on the same app version?

oh-yes-0-fps, 2024-04-09:
Over a month since issue creators reply, this should probably be closed.

Gold872, 2024-04-22:
I was unable to reproduce this issue, feel free to reopen if it happens again or there is more info about what is causing it

closed 2024-04-22 by Gold872