Gold872/elastic_dashboard #80: Multi-topic graphs

DanPeled, 2024-08-02:
ive been trying to implement multi-topic graphs for elastic, but have got into an issue, when i add the other graph it bugs out, i may be using the graph addition wrong, what    i have done it create a list of the FastLineSeriers i add instead of adding one manually, but it seems to be erasing the graph, not sure how to describe that

code :
```dart
for (var topic in _additionalTopics) {
      logger.info(topic);
      List<_GraphPoint> points = handleAdditionalTopics(topic);
      list.add(FastLineSeries<_GraphPoint, num>(
        name: topic,
        animationDuration: 0.0,
        animationDelay: 0.0,
        sortingOrder: SortingOrder.ascending,
        color: const Color.fromARGB(255, 69, 71, 73),
        width: widget.lineWidth,
        dataSource: points,
        xValueMapper: (value, index) => value.x,
        yValueMapper: (value, index) => value.y,
        sortFieldValueMapper: (datum, index) => datum.x,
      ));
    }
    ```

![image](https://github.com/user-attachments/assets/77b0ad59-7ec3-4370-ba40-bc8097a983dd)

closed 2024-08-16 by DanPeled

Zynh0722, 2025-02-13:
hate to dreg up old issues, but is there a reason for this one to be not planned?

Zynh0722, 2025-02-13:
this might be a seperate issue, but even graphs for Number Arrays would work good enough I think

DanPeled, 2025-02-13:
If u ask me, not really, according to @Gold872  It's due to elastic not being designed for graphs and only for drivers, which dont want stuffl like graphs in their dashboard