Gold872/elastic_dashboard #126: Linux Packaging

blaze-developer, 2024-10-28:
Is there any simple way with flutter to package programs for different linux distributions? It would make install so much simpler if we could just use a package file to install the program. (Ex. a arch linux package, and a deb package). If this is simple to implement, it would be a very much appreciated upgrade <3 You have a wonderful project and I look forward to seeing it evolve :3

DanPeled, 2024-11-07:
I was trying to create something like this, but got into some issues when packaging, will probably try again in some time

DanPeled, 2024-11-30:
currently trying to achive this via [this](https://github.com/DanPeled/dashboard-rewrite/blob/feature/linux_packaging/linux_package.sh) but it is only half-working

blaze-developer, 2025-02-13:
Now that Elastic comes with WPILIB, this is less of an issue. But an official package would definitely still be ideal.