Gold872/elastic_dashboard #141: Sendables not Draggable after beta 1

Ryanforce08, 2024-11-27:
I have a sendable builder that used to be draggable by list before beta 1 which introduces widget registrar which i believe is the issue but after that its no longer possible. Here is the code that makes the sendable for example https://github.com/FRC5113/2024-Testing-Swerve/blob/main/magicbot/util/smart_preference.py

DanPeled, 2024-11-27:
I have been using the elastic beta 4 yesterday and it worked, what version of elastic is this currently happening to u on?

Ryanforce08, 2024-11-27:
It has happened on every version after beta 1

DanPeled, 2024-11-28:
Are u able to provide a video of the behavior?

Ryanforce08, 2024-11-29:
https://github.com/user-attachments/assets/ecfaf2b7-a2d1-4e63-9957-cbc09e6fc433

DanPeled, 2024-11-29:
This is really odd, I just tested yesterday and it seems to work, how are u posting that topic?

Ryanforce08, 2024-11-30:
We use sendable builder

DanPeled, 2024-11-30:
is that the case for all the sendables or just the custom ones u made? i see u are not trying to drag the others

Ryanforce08, 2024-11-30:
Custom ones

DanPeled, 2024-11-30:
Then probably ur doing something wrong and not the dashboard

Ryanforce08, 2024-11-30:
Then why would it have worked before beta

DanPeled, 2024-11-30:
extermely unsure, but it is 90% something wrong with how u did it

sciencewhiz, 2024-12-02:
You could try bisecting, but looking through the changes since beta 1, I'd guess this was the change: https://github.com/Gold872/elastic-dashboard/pull/110

Ryanforce08, 2024-12-02:
> extermely unsure, but it is 90% something wrong with how u did it
> 
> Try to run that same code with the latest acutal release and not a beta one and lmk if it works

it does work and tried beta 1, which is when it stops working

Ryanforce08, 2024-12-03:
I was looking through stuff, and I think it has to do with #43, so when I use Builder. setSmartDashboardType("SmartController") is not a built-in widget, which makes it undragable. This is just a theory though

DanPeled, 2024-12-03:
yes, theres a high possibility its that

Gold872, 2024-12-03:
I was finally able to reproduce this issue, what I did was started a server with outline viewer, and published data from an old log file of one of my team's matches. Initially, I noticed that no list layouts were created (none of the `.type` topics were published), however, if I held the drag for a few seconds, the widget would appear. I'm not sure if this is the exact same issue you are having, but I encourage to try holding the drag for a few seconds to see.

With this in mind though, I've come up with the following:
1. From what I tested, I don't believe there were any changes in Beta 1 that would lead to this type of behavior
2. The widget actually does get built, it just takes a long time

The issue is caused by these 2 snippets:
https://github.com/Gold872/elastic-dashboard/blob/1a993f358033874e461d728cd53f676aafd48ce1/lib/widgets/network_tree/networktables_tree_row.dart#L185-L187
By default, it will wait 2.5 seconds to receive a response, if it's a case like mine where there is no `.type` data, then it will wait before continuing, which isn't really necessary

https://github.com/Gold872/elastic-dashboard/blob/1a993f358033874e461d728cd53f676aafd48ce1/lib/widgets/network_tree/networktables_tree_row.dart#L204-L223
When creating the list layout, it will create each widget one at a time instead of doing them all at once, which further increases the list layout creation time

To fix this, I'm going to wait for all of the list layout widgets to build at once, so multiple widgets could be "fetched" at once

Regarding the `.type` issue, I can start by decreasing the timeout period, however it could also lead to "false positives" where the timeout is too low to receive any data. I'll continue looking into the solution to the timeout problem.

Try reproducing this again but instead, hold the drag for much longer and see if the widget appears

Ryanforce08, 2024-12-04:
The widget did not appear sadly. I m going to look through code and see if I can figure what changed in beta 1

Gold872, 2024-12-04:
Could you also send the code you are using to register this custom widget? It's possible there's an issue on that end

Ryanforce08, 2024-12-04:
Here you go. This is the file that it's in https://github.com/FRC5113/2024-Testing-Swerve/blob/main/magicbot/util/smart_preference.py

Gold872, 2024-12-07:
Found the issue, and it was a weird one

https://github.com/Gold872/elastic-dashboard/pull/151

Essentially what was happening was the widget type wasn't registered, so it created a "Stub" widget (used for widgets without custom properties, and since it's not a coded in widget, it wouldn't build a widget and cancel the operation. Since it was able to create a stub widget, it chose not to create a list layout, which is why nothing would be created.

Not sure what was changed after beta 1 that broke it, I'm surprised it worked before that

PR #151 (2024-12-07): Fix list layout creation for unregistered sendable types
- Allows sendables with an unknown/unregistered type to be turned into a list layout
- Speeds up creation of list layouts
- Fixes an issue with list layouts not unsubscribing if a drag finishes before list layout is created

Resolves #141
commit 8b9f7847 (2024-12-07): Fix list layout creation for unregistered sendable types (#151)

- Allows sendables with an unknown/unregistered type to be turned into a
list layout
- Speeds up creation of list layouts
- Fixes an issue with list layouts not unsubscribing if a drag finishes
before list layout is created

Resolves #141
closed 2024-12-07 by Gold872