Gold872/elastic_dashboard #196: Default window size is broken on macOS

jwbonner, 2025-01-14:
**Describe the bug**
This isn't strictly related to #195, but was something I noticed when experimenting with window sizes. The default window size (at least on macOS) appears to be smaller than the minimum window size and results in overlapping UI elements. Resizing the window at all snaps it to the (much larger) minimum window size.

[image]
 
**To Reproduce**
Steps to reproduce the behavior:
1. Disable "Remember Window Position"
2. Open Elastic

**Version:**
* Elastic 2025.0.2
* macOS 15.2

Gold872, 2025-01-14:
Out of curiosity, what's your display resolution and DPI? The calculation for the minimum size was very scrapped together (and doesn't even account for DPI) so I'm not surprised that it causes weird behaviors like this

jwbonner, 2025-01-15:
The native resolution of the screen is 3024x1964, but I have it scaled to 1512x982. I can do some tests at other resolutions later.

Gold872, 2025-01-31:
This should have been automatically closed by #197

closed 2025-01-31 by Gold872