Gold872/elastic_dashboard #198: Linux: app icon is not properly set

crueter, 2025-01-15:
**Describe the bug**
On Linux, Elastic does not have an associated app icon on the app launcher, resulting in it showing up as the generic Xorg icon instead. (untested on Wayland)

**To Reproduce**
Steps to reproduce the behavior:
1. Open Elastic on Linux, be it WPILib or the official release

**Expected behavior**
The application launcher should display the Elastic or WPILib logo.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Gentoo Linux (rolling)
 - Desktop: KDE Plasma 6.2.4

**Additional context**
I'm fairly confident the solution to this lies in explicitly setting the app's icon in code. I have no idea how to do this with Flutter, but the Qt equivalent would be:

```
app.setWindowIcon(QIcon(":/iconName"));
```

Gold872, 2025-01-15:
If you go to the application file itself, does it have the proper icon? Or is it just an issue with the icon being displayed while running the app?

Sneedplex, 2025-01-15:
Yes, the WPILib-provided desktop file has the proper WPILib icon. From what I can tell, StartupWMClass is needed for Wayland, and setting the window icon is needed for X11.

crueter, 2025-01-15:
Yes, the WPILib-provided desktop file contains the WPILib icon. As far as I can tell, Wayland necessitates StartupWMClass and X11 necessitates setting the window icon from the app itself. The binary file doesn't have the icon, it has the standard octet-stream icon.

[image]

[image]

crueter, 2025-01-16:
That PR actually fixed both, no code changes needed.