Gold872/elastic_dashboard #199: Setting to automatically enable submit button

glolichen, 2025-01-17:
First of all, I really appreciate this app. It is **a lot** better than Shuffleboard/SmartDashboard.

We found it was somewhat time consuming to add a submit button to every text input, especially when tuning PID or similar.

It would be really great if there were a global setting to
- ***automatically add a submit button to every text widget created***
- ***automatically submit***

Looking around past issues I understand from #175 not submitting automatically is intended, but some teams might want it to automatically submit. Adding an option gives teams that option.

I understand from that PR/Issue and #174 that there are options that sort of do this kind of thing by automatically enabling submit buttons for persistent entries. This is somewhat distinct from that.

Thank you!

Gold872, 2025-01-18:
> Looking around past issues I understand from https://github.com/Gold872/elastic-dashboard/pull/175 not submitting automatically is intended, but some teams might want it to automatically submit. Adding an option gives teams that option.

This has been intentionally left out, since even if the user knows that it will automatically submit, there's still a risk to safety, as there could be a typo, or a possible miscommunication that can cause unexpected behavior. If there's ever a time when values are being updated while the robot is enabled, I think it should be an explicit action from the user.

> I understand from that PR/Issue and https://github.com/Gold872/elastic-dashboard/issues/174 that there are options that sort of do this kind of thing by automatically enabling submit buttons for persistent entries. This is somewhat distinct from that.

In most cases, a topic used for PID values should be persistent, as its intended to be a "tune once" type of value where it is saved across code restarts. I'm not entirely opposed to adding this but I'm not sure how necessary of an addition it is. Also, Elastic has a PID controller widget, which I've heard people have found very useful, not sure if that's what you're looking to use though.

glolichen, 2025-01-19:
This adds a setting under developer which is off by default. I think the logic/implementation works? This isn't that important of a feature but I'm sure some people would appreciate having more options (including my team). Thanks for the time.

closed 2025-02-06 by glolichen