Gold872/elastic_dashboard #214: Elastic using lots of ram

me-it-is, 2025-02-05:
Elastic is using 4-12 gb of ram

1. run [https://github.com/ChainLynxRobotics/2025_Season/tree/drivetrain-vision](url)

using less ram

Windows 11

sciencewhiz, 2025-02-05:
Does it use that much ram when you run your code in simulation? I only saw it use 1.6-1.8gb when I ran your code in simulation

Gold872, 2025-02-05:
I was doing some testing with your code, and it looks like you're sending a ton of notifications every single robot loop, which is clogging up the Flutter renderer and causing all sorts of issues in the program. I'll try to look into ways of fixing it, but in the meantime I suggest not sending so many notifications. The non-persistent notifications are meant for one-time messages that will appear temporarily. For frequent errors or warnings like this, I'd recommend not even sending it to the dashboard, as it will all be sent to the driver station and stored in logs.

me-it-is, 2025-02-08:
thanks i didn't realize that it was sending it every loop it was only supposed to send a notification when it changes.

closed 2025-02-08 by me-it-is