Gold872/elastic_dashboard #217: Crashing on opening SendableChooser

zeroClearAmerican, 2025-02-11:
**Describe the bug**
We have implemented a custom SendableChooser in our robot's Java code. The implementation is nearly identical to the original, with some added functions for managing the internal map of options available in the chooser. In certain situations we're observing that the Elastic widget for the chooser suddenly clears the default/selected option, and then freezes/becomes unresponsive when trying to open the combo box. 

[Here](https://github.com/FRCTeam31/2025_competition_robot/blob/pathplanner_fixes2/src/main/java/org/prime/dashboard/PrimeSendableChooser.java) is the custom sendable chooser implementation. 

When this happens, we can still see proper selected/default items, and valid options in the network tables tree for the chooser, but the widget acts as if there are no options, and nothing selected, and then freezes the app entirely, requiring a restart to resolve.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the "pathplanner-fixes2" branch of the robot code from [here](https://github.com/FRCTeam31/2025_competition_robot/tree/pathplanner_fixes2).
2. Open Elastic
3. Run the project in simulation mode, using the Sim GUI.
4. Download the Elastic layout from the robot.
5. Go to the "Auto" tab.
6. Using the `Auto Options` chooser, select "S2-to-IJ", then press `Add Step` below the chooser. The step should be added in `Routine Steps`.
7. Now select "IJ-to-SRC2", then press `Add Step`. The step should be added in `Routine Steps`.
8. Now select "SRC2-to-IJ", then press `Add Step`. The step should be added in `Routine Steps`, but the chooser should now no longer have an option selected.
9. Click the `Auto Options` dropdown, which will cause Elastic to freeze.

**Expected behavior**
Trying to select options from a chooser should not cause the application to freeze.

**Screenshots**
[image]
(just before the crash)

[image]
(right after clicking the `Auto Options` dropdown from the first screenshot)

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 25.0.2

Gold872, 2025-02-12:
I was able to reproduce this issue as described. It looks like what's happening is when `Add Step` is pressed, the step that is added is removed from Auto Chooser. Flutter expects that the current selection for a dropdown chooser is one of the available options, and when it isn't, it will throw a runtime error and display a gray square over the widget. The WPILib SendableChooser always meets this condition, which is why it's usually safe to build the dropdown chooser.

There's a few ways I would recommend fixing this:
1. Clear the selected option when adding a step
2. Don't remove any of the options, and instead check whether or not the step is already present when trying to add a step

This is a unique way of making autos and I'm really impressed by it!

zeroClearAmerican, 2025-02-13:
Thanks for pointing that out! Makes sense. I was able to fix the problem armed with that knowledge. [Here's the working branch](https://github.com/FRCTeam31/2025_competition_robot/tree/autoroutine-paths-and-history) if you'd like to take a look.

For future reference, does Elastic save an application log that could be useful in tracking down the source of an error or crash? Apologies if this is a dumb question; Flutter is not my bread and butter. 

> This is a unique way of making autos and I'm really impressed by it!

Thanks! We're a smaller team so hopefully - with some solid QA and driver training of course - this system will greatly reduce the work required to create and maintain separate PathPlanner autos for every possible combination of paths that can be run.

closed 2025-02-13 by zeroClearAmerican