Gold872/elastic_dashboard #235: Using Elastic.java to switch tabs causes memory leak and crash

HugoKlepsch, 2025-03-05:
**Describe the bug**
A clear and concise description of what the bug is.

We added Elastic to our robot code because we wanted to switch Elastic tabs when moving from autonomous to teleoperated control. The diff for this change is roughly this: 

Add `Elastic.java`, copied from [elasticlib/Elastic.java](https://github.com/Gold872/elastic-dashboard/blob/main/elasticlib/Elastic.java) in this repo. 
Add tab switching code to `teleopInit()`, so that it only runs once, as directed by [this doc](https://frc-elastic.gitbook.io/docs/additional-features-and-references/robot-notifications-with-elasticlib#tab-switching):

```
diff --git a/src/main/java/frc/robot/Robot.java b/src/main/java/frc/robot/Robot.java
index 93a58c7..d55ca72 100644
--- a/src/main/java/frc/robot/Robot.java
+++ b/src/main/java/frc/robot/Robot.java
@@ -114,6 +114,7 @@ public class Robot extends TimedRobot {
   @Override
   public void autonomousInit() {
+    Elastic.selectTab("Auto");
 
@@ -136,6 +137,7 @@ public class Robot extends TimedRobot {
   @Override
   public void teleopInit() {
+    Elastic.selectTab("Teleoperated");
   }
``` 

After adding this code, we noticed that the robot code crashes and restarts roughly every 2 minutes. Reverting the above change the robot code never crashes. I suspect that there is a memory leak in `Elastic.java`. The error logged during the crash indicates that the robot's Java process runs out of memory, or is unable to allocate more memory.

Our code has more allocations than the skeleton starter code, so it might take longer unless on a similarly memory-constrained environment. 

We use the roboRio 1 with 256MB of ram. 

**To Reproduce**
Steps to reproduce the behavior:
(See above)
1. Create Elastic dashboard with tabs
2. Create TimedRobot skeleton
3. Add sample code to switch between tabs
4. Deploy, wait several minutes
5. Robot code crashes with JVM out of memory

**Expected behavior**
No memory leak

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win 11, 2025 driver station, 2025.3.1 WPIlib & bundled Elastic
 - Elastic.java from https://frc-elastic.gitbook.io/docs/additional-features-and-references/robot-notifications-with-elasticlib#installing-elasticlib

Gold872, 2025-03-05:
It sounds like the RIO 1 is _just_ reaching its memory limits within your code, which has been a somewhat common issue in recent years. ElasticLib only creates 2 new publishers in total, which only allocates memory once on startup, but it could be the last little bit of memory allocation causing it to run out.

I would take a look at WPILib's docs on reducing memory usage, and also try some of the other recommendations on this page of the documentation: https://docs.wpilib.org/en/stable/docs/software/basic-programming/java-gc.html#system-memory-tuning

HugoKlepsch, 2025-03-08:
> the RIO 1 is just reaching its memory limits within your code

This is true; we're on the edge. We have done some intial steps in optimizing our memory usage and reducing allocations. We have so little free memory in general, that we're having trouble running the robot code with JVM memory profilers. Connecting with the memory analyzer OOMs the process. If we succeed analyzing the memory and find a leak source in Elastic/NT, I'll update this issue. 

I still think that Elastic has some sort of memory leak; 
- Running without Elastic, the driverstation reports ~20MB free memory, but when using Elastic we have much less (~5MB), and it seems to reduce before causing performance issues and then OOMing. 
- Elastic allocations are supposed to be static, but memory usage climbs while running.  
The leak might be internal to the NetworkTable publishers and not in Elastic.

Gold856, 2025-03-13:
The root cause is the use of Jackson for JSON serialization for Elastic notifications. Merely loading the classes needed for Jackson has been known to use up a bunch of RAM. A temporary fix would be to take out all code related to Elastic notifications and ObjectMapper.

HugoKlepsch, 2025-03-19:
We added some swap space this week, and it seems to have improved the stability of our code and reduced variance in robot code loop durations. I dismissed this option before, because I know how slow swap space is to access. Despite this, it seems to work much better. Maybe it is swapping out less used pages. We're also able to run the VisualVM JVM profiler. I'll try to run Elastic again and compare heap dumps.