Gold872/elastic_dashboard #246: allow Elastic to use a different port for the webserver

tytytyty74, 2025-03-19:
**Is your feature request related to a problem? Please describe.**
The port that elastic uses for remote layout downloading is hardcoded to 5800, which can conflict with other services, such as PhotonVision's web dashboard.

**Describe the solution you'd like**
I would like to change the port that elastic is listening to for remote layout downloading.

**Describe alternatives you've considered**
Currently, the solution we are using is to just not use the remote layout downloading, but it is a useful feature that we would like to use.

**Additional context**
N/A

tytytyty74, 2025-03-19:
yes, when I try to download the layout from the robot, elastic gives an error saying "failed to retrieve layout list, connection closed before full header was recieved. for accessing the web dashboard through the usb cable, photon vision reccomends port forwarding to port 5800 [here](https://docs.photonvision.org/en/latest/docs/quick-start/networking.html#port-forwarding), which are doing for debugging at competition

rzblue, 2025-03-20:
Since the new radios have 4 ethernet ports, I would recommend tethering via ethernet at competitions, and remove port forwarding through the roboRIO.